SINTEF-9012/JArduino #33: Error adding to eclipse project

coalman321, 2015-05-10:
I have added the library to my eclipse session but when I use code form the examples it says that there is nothing to import. no extra imports to resolve the errors. any help would be appreciated because I need this project finished at the end of the week.
Cole

brice-morin, 2015-05-11:
Seems to work. [Download the zip](https://github.com/SINTEF-9012/JArduino/releases/download/JArduino-0.1.7/JArduino-0.1.7.zip) and import it in Eclipse:

- Import...
- Existing projects into Workspace
- Select archive file: <directly point to the zip you just downloaded>

It should create a project in your workspace that is pre-configured

coalman321, 2015-05-11:
I'll give you a breakdown of what i did
First I added it to the project compilers I think. It was the only page you can add external jars
I then began writing the code in from some of the examples. But when I tried to extend the class it could not find a jarduino to extend
I am using the latest version
Thanks

brice-morin, 2015-05-11:
You need to add the `org.sintef.jarduino.core-0.1.7-SNAPSHOT.jar` to your build path (right click on your project -> Build Path -> Configure Build Path... -> Libraries tab)

coalman321, 2015-05-11:
according to eclipse the lirary has already been added
 here is my code as of today

package ardubot;
import org.sintef.jarduino.JArduino;

public class JComm extends JArduino{
	
	 public JComm(String port) {
		    super(port);

	
	  @Override
	    protected void setup() {
		  
		  
		  
	  }
	
	  @Override
	    protected void loop() {
		  
		  
		  
	  }

	
	public static void main(String[] args){
		  JArduino arduino = new JComm("COM3");
	        arduino.runArduinoProcess();

	}

}

brice-morin, 2015-05-11:
Then try the first option I propose, it will deploy a project that is pre-configured and containing some samples. Then you can simply edit one of the existing sample, or create a new one in this project

coalman321, 2015-05-11:
there is only one problem i already have two or three classes in this project that are necessary for this "applet" to function

also how do you do analog write in jarduino?
thanks

brice-morin, 2015-05-11:
You can do analog write as follows:
https://github.com/SINTEF-9012/JArduino/blob/master/jarduino.samples/src/main/java/org/sintef/jarduino/examples/analog/AnalogInOutSerial.java

coalman321, 2015-05-11:
where is the jar because i have just been adding the zip
this ma have been the problem in the first place because when i do the import org.sintef.jarduino.JArduino the org.sintef part is underlined

brice-morin, 2015-05-11:
If you follow the procedure:

- right click in your workspace
- Import...
- Existing projects into Workspace
- Select archive file: point to the zip you have downloaded [here](https://github.com/SINTEF-9012/JArduino/releases/download/JArduino-0.1.7/JArduino-0.1.7.zip)

It should create a project in your workspace that is pre-configured, with the build path properly configured and pointing to the jars, etc, and a set of samples that compiles

coalman321, 2015-05-11:
i started with a clean slate
i attempted to import it without a project already started, it did not want to import
created a new project and imported the project into the workspace. the folder shows up as a folder not a project folder. i created a new class in the project folder i had created to import the library. I tried to reinsert the code and update it to the correct package and class name
the same problem occurs, it cannot find the jarduino import

 here is a screenshot
[image]

brice-morin, 2015-05-11:
open the JArduino-0.1.7 folder and nagivate to the jar. Then right click on the jar add add to build path

coalman321, 2015-05-11:
it works 
thank you so much for your help i will now close the thread

brice-morin, 2015-05-11:
No problem. Just remember to star the project if you liked it :-)

coalman321, 2015-05-11:
i just got around to finishing an creating the code. my arduino is at home but i ran the code and this appeared in the system line

here it is
Load RxTx
java.lang.UnsatisfiedLinkError: no rxtxSerial in java.library.path thrown while loading gnu.io.RXTXCommDriver
Exception in thread "main" java.lang.UnsatisfiedLinkError: no rxtxSerial in java.library.path
	at java.lang.ClassLoader.loadLibrary(Unknown Source)
	at java.lang.Runtime.loadLibrary0(Unknown Source)
	at java.lang.System.loadLibrary(Unknown Source)
	at gnu.io.CommPortIdentifier.<clinit>(CommPortIdentifier.java:123)
	at org.sintef.jarduino.comm.Serial4JArduino.connect(Serial4JArduino.java:88)
	at org.sintef.jarduino.comm.Serial4JArduino.<init>(Serial4JArduino.java:82)
	at org.sintef.jarduino.AbstractJArduino.<init>(AbstractJArduino.java:30)
	at org.sintef.jarduino.JArduino.<init>(JArduino.java:45)
	at ardubot.JComm.<init>(JComm.java:12)
	at ardubot.JComm.jcom(JComm.java:48)
	at ardubot.Main.main(Main.java:6)
 i dont know whether this has to do with the arduino not being connected or it is an actual error

dosjos, 2015-05-11:
Yeah, it could mot find any devices connected to a serial port. Connecting an arduino should do The trick

at gnu.io.CommPortIdentifier.(CommPortIdentifier.java:123)
at org.sintef.jarduino.comm.Serial4JArduino.(Serial4JArduino.java:82)
at org.sintef.jarduino.AbstractJArduino.(AbstractJArduino.java:30)
at org.sintef.jarduino.JArduino.(JArduino.java:45)
at ardubot.JComm.(JComm.java:12)

coalman321, 2015-05-11:
phew it looked like it but because it warned about RXTX i was slightly worried

coalman321, 2015-05-11:
OK one last time. i went to compile the firmware and load to the arduino. i opened the firmware.pde and hit compile. it began to compile until it hit line 22

JArduinoFirmware.pde:18:22: fatal error: JArduino.h: No such file or directory
compilation terminated.

the jarduino.h is not located in the same folder however it is right outside. Im guessing i need to put it somewhere.
thanks

coalman321, 2015-05-12:
Ok i wnrt in and began to modernize the code as suggested. i moved jarduino.cpp and jarduino.h i then reconfigured the files to their updated states. now i get a similar error

Arduino: 1.6.3 (Windows 8.1), Board: "Arduino Uno"

JArduinoFirmware.ino:20:22: fatal error: JArduino.h: No such file or directory

compilation terminated.

Error compiling.

same error different line don't know what to do now. any ideas?

coalman321, 2015-05-12:
ok i got through the firmware issues back to the software issues. the error i previously received about the RXTX library keeps happening no matter what port i specify. port 3 is default but according to the arduino ide it says port 4

brice-morin, 2015-05-12:
Then you should change to `"COM4"` in your main

coalman321, 2015-05-12:
ok i  updated it to "COM4" which the Arduino IDE identifies. i then run the java program ang get the same error. here is the error, it may have changed.

Load RxTx
java.lang.UnsatisfiedLinkError: no rxtxSerial in java.library.path thrown while loading gnu.io.RXTXCommDriver
Exception in thread "main" java.lang.UnsatisfiedLinkError: no rxtxSerial in java.library.path
	at java.lang.ClassLoader.loadLibrary(Unknown Source)
	at java.lang.Runtime.loadLibrary0(Unknown Source)
	at java.lang.System.loadLibrary(Unknown Source)
	at gnu.io.CommPortIdentifier.<clinit>(CommPortIdentifier.java:123)
	at org.sintef.jarduino.comm.Serial4JArduino.connect(Serial4JArduino.java:88)
	at org.sintef.jarduino.comm.Serial4JArduino.<init>(Serial4JArduino.java:82)
	at org.sintef.jarduino.AbstractJArduino.<init>(AbstractJArduino.java:30)
	at org.sintef.jarduino.JArduino.<init>(JArduino.java:45)
	at ardubot.JComm.<init>(JComm.java:12)
	at ardubot.JComm.jcom(JComm.java:48)
	at ardubot.Main.main(Main.java:6)

also here i s the curent code

package ardubot;
import org.sintef.jarduino.JArduino;
import org.sintef.jarduino.PWMPin;

public class JComm extends JArduino{
	
	public static int arow;
	 final PWMPin lm = PWMPin.PWM_PIN_11;
	 final PWMPin rm = PWMPin.PWM_PIN_10;
	
	 public JComm(String port) {
		    super(port);
	 }
	
	  @Override
	    protected void setup() {
		   
	  }
	
	  @Override
	    protected void loop() {
		  
		if(arow == 1){
			analogWrite(lm, (byte) 225);
			analogWrite(rm, (byte) 250);
		}
		if(arow == 2){
			analogWrite(lm, (byte) 140);
			analogWrite(rm, (byte) 125);
		}
		if(arow == 3){
			analogWrite(lm, (byte) 125);
			analogWrite(rm, (byte) 250);
		}
		if(arow == 4){
			analogWrite(lm, (byte) 250);
			analogWrite(rm, (byte) 125);
		}
		else{
			analogWrite(rm, (byte) 185);
			analogWrite(lm, (byte) 185);
		}
		  
	  }

	
	public static void jcom(){
		  JArduino arduino = new JComm("COM4");
	        arduino.runArduinoProcess();

	}
[... 834 lines cut ...]
    72  ldc <String "linux"> [68]
    74  invokevirtual java.lang.String.indexOf(java.lang.String) : int [69]
    77  iconst_m1
    78  if_icmpne 81
    81  ldc <String "rxtxSerial"> [70]
    83  invokestatic java.lang.System.loadLibrary(java.lang.String) : void [71]
    86  return
      Exception Table:
        [pc: 10, pc: 28] -> 31 when : java.lang.Throwable
      Line numbers:
        [pc: 0, line: 104]
        [pc: 10, line: 107]
        [pc: 22, line: 108]
        [pc: 28, line: 113]
        [pc: 31, line: 110]
        [pc: 32, line: 112]
        [pc: 62, line: 117]
        [pc: 68, line: 118]
        [pc: 81, line: 123]
        [pc: 86, line: 124]
}

with this at the top

[image]

brice-morin, 2015-05-12:
What is your OS?

Can you modify [this file](https://github.com/SINTEF-9012/JArduino/blob/master/jarduino.serial/src/main/java/org/sintef/jarduino/comm/Serial4JArduino.java) and print your osName and osProc. It seems that the DLL (or SO) files needed for your OS is not deployed.

coalman321, 2015-05-12:
i am using windows 8.1 
do you want me to put it inside of my current project?
also what do i need to modify

brice-morin, 2015-05-12:
Yes, put this code in your main:

```java
String osName = System.getProperty("os.name");
String osProc = System.getProperty("os.arch");
System.out.println("osName: " + osName);
System.out.println("osProc: " + osProc);
```

And copy/paste the osName and osProc

coalman321, 2015-05-12:
ok here is what the console put out
osName: Windows 8.1
osProc: amd64

brice-morin, 2015-05-12:
OK, I think I fixed the problem but cannot make a new release yet. You'll have to use JArduino directly by pulling the sources from the repository. Make sure you have Maven installed and properly configured.

See https://github.com/SINTEF-9012/JArduino#serial-usage

coalman321, 2015-05-12:
where does this new .java need to go?

coalman321, 2015-05-13:
i dont have jarduino.serial however i have the rxtx library installed do i need to exchange them?

closed 2015-05-16 by coalman321