SINTEF-9012/JArduino #34: Example: Compiling and Running general JArduino class

blurredHorizon, 2015-05-16:
I'm trying to get JArduino to work but I'm having trouble running the examples.  I apologize for not being able to figure this out on my own.  

/*background: 
--I have confirmed that my board is in good working order and serial communication using the native Arduino language works.
--I've never used Maven, so I don't understand how to create and run a Java class file from it.
--I'm using Ubuntu 14.04 and an UNO R3
*/  

I successfully compiled and uploaded the JArduino firmware sketch to my board.

I have taken the following steps in order to compile and run the blink example:
1. Sudo apt-get install maven2
2. cd JArduino-0.1.7/jarduino.samples/
(The readme says to do step 3 under JArduino, but the POM file is in this jarduino.samples directory so I'm guessing that's where it needed to be done)
3. mvn clean install
4. mvn exec:java -Dexec.mainClass="org.sintef.jarduino.examples.basic.Blink"

Considering that I can't even get the example to function I was wondering if someone could detail the steps to getting a general JArduino class to compile and run.

Thanks!

Cookiez-, 2015-09-05:
I haven't used Maven but I am using Eclipse and linking libraries to my projects. I can guide you using this method but maybe Maven is similar. 
https://github.com/SINTEF-9012/JArduino/wiki/2-Minute-Tutorial

blurredHorizon, 2015-12-11:
Great!  I'll try this out!  Thanks.

aliassad, 2016-04-14:
I successfully compiled and uploaded the JArduino firmware to my board.
but i am unable to run and use jarduino libaray in eclipse facing to many error but nothing done.
Kindly create a simple tutorial to run and use jarduino library with picture.

thanks in advance