SINTEF-9012/JArduino #35: unable to run blink example

jerryd84, 2015-08-08:
I've run the firmware example in Arduino on my uno board.
I've created the default project with the jars and libraries preconfigured in eclipse
I edited the blink example to output on pin 13.

when I run the example i get the following message on the console
"Load RxTx"
And I'm prompted to select the comport of my Arduino which for me is 10.
After selecting the comport, the Rx light blinks continuously and the LED on pin 13 remains lit indefinitely.

Please advise.
Thank you.

brice-morin, 2015-08-10:
Hi,

For some reasons, the blink example does not use PIN 13 but assumes that you have a LED connected on PIN 12. See the [source code](https://github.com/SINTEF-9012/JArduino/blob/master/jarduino.samples/src/main/java/org/sintef/jarduino/examples/basic/Blink.java)

Try changing the PIN number to PIN 13 in the program, so that it should bling the embedded LED on the Arduino board.

Please tell us if it solve the problem, or not.

jerryd84, 2015-08-12:
I did change the output pin to 13. I also uploaded the firmware and setup the libraries in eclipse. I ran into the mismatch of libraries but I fixed it and no longer see the mismatch error. When I run the program the 
Rx light continuously blinks and the led pin on 13 just remains on.

DoctorPainless, 2018-01-28:
I spend some hours getting Jardiuno up and runnig and after installing the correct (in my case x64) RTXT .dll files (google for rxtx-for-java) it worked.