SINTEF-9012/JArduino #39: fatal error: WConstants.h: No such file or directory

frickerm, 2015-11-10:
Hello
I have got some problems with upload the firmware to my Arduino

I have downloadet the JArduino-0.1.7 zip file and copy the arduino folder to the libery by the Arduino on my Computer.

Then i oppened the Arduino ide 1.6.6 (windows 64 bit)

I connected the Arduino UNO and launch the ide.

on File> Examples> Jarduino> JArduino Firmware.

the programm started... but if i compile and want to upload this error will be shown:

Arduino: 1.6.6 (Windows 8.1), Board: "Arduino/Genuino Uno"

C:\Program Files (x86)\Arduino\libraries\JArduino/JArduino.h:24:24: fatal error: WConstants.h: No such file or directory

 #include "WConstants.h"

                        ^

compilation terminated.

Bibliothek JArduino im Ordner: C:\Program Files (x86)\Arduino\libraries\JArduino (legacy) wird verwendet
Bibliothek EEPROM in Version 2.0 im Ordner: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\EEPROM  wird verwendet
exit status 1
Fehler beim Kompilieren.

brice-morin, 2015-11-10:
Did you download version 1.7 or 1.7b?

1.7b should fix the problem. You need to include Arduino.h instead of WConstants.h.

However, it might be that Arduino changed the way they deal with version number and that piece of code might not work anymore:

```
//JArduino.h
#if defined(ARDUINO) && ARDUINO >= 100
	#include "Arduino.h"
#else
	#include "WConstants.h"
#endif
```

```
//JArduino.cpp
#if defined(ARDUINO) && ARDUINO >= 100
	#include "Arduino.h"
#else
	#include "WProgram.h"
#endif
```

If you did download v1.7b, simply include Arduino.h in your JArduino.h and JArduino.cpp and see if it helps.

brice-morin, 2015-11-10:
v1.7b compiles with Arduino 1.6.6

closed 2015-11-10 by brice-morin

frickerm, 2015-11-10:
it works thanks