SINTEF-9012/JArduino #40: Jarduino blink loop  start with an button

frickerm, 2015-11-17:
Hi 
i have got some problems..

i downloaded the JArduino file and installed all.
the blink example and the others work.

My problem is 
I want to connect the blink example with an Java -GUI 

I build the gui and a button. with this button i want to start the loop methode in the blink example (in my projekt calls Auto_Test_Folge)

but an error : change Auto_Test_Folge.loop() to an static methode.
if i change this it dont works 

heare the Source code

GUI

		     btnAutotest.addActionListener(new ActionListener() {
			    	public void actionPerformed(ActionEvent arg0) {
			    		
			    		
			    	Auto_Test_Folge.loop();
			    	}
			    });

Auto_test_Folge

import org.sintef.jarduino.DigitalPin;
import org.sintef.jarduino.DigitalState;
import org.sintef.jarduino.JArduino;
import org.sintef.jarduino.PinMode;
import org.sintef.jarduino.comm.Serial4JArduino;
/*
Blink
Turns on an LED on for one second, then off for one second, repeatedly.

This example code is in the public domain.
 */

public class Auto_Test_Folge extends JArduino {

    public Auto_Test_Folge(String port) {
        super(port);
        loop();
    }

    public void setup() {
        // initialize the digital pin as an output.
        // Pin 13 has an LED connected on most Arduino boards:
        pinMode(DigitalPin.PIN_10, PinMode.OUTPUT);
        pinMode(DigitalPin.PIN_9, PinMode.OUTPUT);
        pinMode(DigitalPin.PIN_8, PinMode.OUTPUT);
        pinMode(DigitalPin.PIN_7, PinMode.OUTPUT);
        pinMode(DigitalPin.PIN_6, PinMode.OUTPUT);
        pinMode(DigitalPin.PIN_5, PinMode.OUTPUT);
        pinMode(DigitalPin.PIN_4, PinMode.OUTPUT);
        pinMode(DigitalPin.PIN_3, PinMode.OUTPUT);
        pinMode(DigitalPin.PIN_2, PinMode.OUTPUT);

        
    }

    public void loop() {
    	   // Schaltet nach der reihe die LEDs auf den Pins an
        digitalWrite(DigitalPin.PIN_10, DigitalState.HIGH);
        delay(1000); // Zeit Einstellung
   
        digitalWrite(DigitalPin.PIN_9, DigitalState.HIGH);
        delay(1000); // Zeit Einstellung
    
        digitalWrite(DigitalPin.PIN_8, DigitalState.HIGH);
        delay(1000); //Zeit Einstellung

        digitalWrite(DigitalPin.PIN_7, DigitalState.HIGH);
        delay(1000); //Zeit Einstellung
    
        digitalWrite(DigitalPin.PIN_6, DigitalState.HIGH);
        delay(1000); // Zeit Einstellung

        digitalWrite(DigitalPin.PIN_5, DigitalState.HIGH);
        delay(1000); // Zeit Einstellung

        digitalWrite(DigitalPin.PIN_4, DigitalState.HIGH);
        delay(1000); //Zeit Einstellung

        digitalWrite(DigitalPin.PIN_3, DigitalState.HIGH);
        delay(1000); // Zeit Einstellung

        digitalWrite(DigitalPin.PIN_2, DigitalState.HIGH);
        delay(1000); // Zeit Einstellung
        
        
        // Schalet LEDs wieder AUS 
        digitalWrite(DigitalPin.PIN_10, DigitalState.LOW);
        delay(1000); // Zeit Einstellung

        digitalWrite(DigitalPin.PIN_9, DigitalState.LOW);
        delay(1000); //Zeit Einstellung

        digitalWrite(DigitalPin.PIN_8, DigitalState.LOW);
        delay(1000); // Zeit Einstellung

        digitalWrite(DigitalPin.PIN_7, DigitalState.LOW);
        delay(1000); //Zeit Einstellung
        
        digitalWrite(DigitalPin.PIN_6, DigitalState.LOW);
        delay(1000); // Zeit Einstellung

        digitalWrite(DigitalPin.PIN_5, DigitalState.LOW);
        delay(1000); // Zeit Einstellung

        digitalWrite(DigitalPin.PIN_4, DigitalState.LOW);
        delay(1000); // Zeit Einstellung

        digitalWrite(DigitalPin.PIN_3, DigitalState.LOW);
        delay(1000); //Zeit Einstellung
 
        digitalWrite(DigitalPin.PIN_2, DigitalState.LOW);
        delay(1000); // Zeit Einstellung
        

        System.exit(0); //Beendet die Schleife und das Programm
    }

	public static void main(String[] args) {
        String serialPort;
        if (args.length == 1) {
            serialPort = args[0];
        } else {
            serialPort = Serial4JArduino.selectSerialPort();
        }
        JArduino arduino = new Auto_Test_Folge(serialPort);
        arduino.runArduinoProcess();
    }

	}

dosjos, 2015-11-17:
In your code change it to

 btnAutotest.addActionListener(new ActionListener() {
                public void actionPerformed(ActionEvent arg0) {
                 Auto_Test_Folge atf = new Auto_Test_Folge();

                atf.loop();
                }
            });

this will of course only run loop once.

-- 
Jan-Ole

frickerm, 2015-11-17:
Ok but now the complete program because of the System.exit(0) in the void methode.
This should not happen it only should stop the methode and close the Auto_Test_Folge

    // Schalet LEDs wieder AUS 
    digitalWrite(DigitalPin.PIN_10, DigitalState.LOW);
    delay(1000); // Zeit Einstellung

    digitalWrite(DigitalPin.PIN_9, DigitalState.LOW);
    delay(1000); //Zeit Einstellung

    digitalWrite(DigitalPin.PIN_8, DigitalState.LOW);
    delay(1000); // Zeit Einstellung

    digitalWrite(DigitalPin.PIN_7, DigitalState.LOW);
    delay(1000); //Zeit Einstellung

    digitalWrite(DigitalPin.PIN_6, DigitalState.LOW);
    delay(1000); // Zeit Einstellung

    digitalWrite(DigitalPin.PIN_5, DigitalState.LOW);
    delay(1000); // Zeit Einstellung

    digitalWrite(DigitalPin.PIN_4, DigitalState.LOW);
    delay(1000); // Zeit Einstellung

    digitalWrite(DigitalPin.PIN_3, DigitalState.LOW);
    delay(1000); //Zeit Einstellung

    digitalWrite(DigitalPin.PIN_2, DigitalState.LOW);
    delay(1000); // Zeit Einstellung

    System.exit(0); //Beendet die Schleife und das Programm
}

dosjos, 2015-11-17:
Just remove the System.Exit statement. When the code reaches the end, it
jumps out of the loop method and you are back actionPerformed again

frickerm, 2015-11-17:
but it don´t work it will be jump up on the top of the methode loop()

here ist the code 

Gui
	
		     //btnAutotest Button Funktion
		     btnAutotest.addActionListener(new ActionListener() {
			    	public void actionPerformed(ActionEvent arg0) {
			    		 Auto_Test_Folge atf = new Auto_Test_Folge(null);

			                atf.loop();
			    	}
			    });
		     

Auto_Test_Folge

public class Auto_Test_Folge extends JArduino {

    public Auto_Test_Folge(String port) {
        super(port);
        loop();
    }

    public void setup() {
        // initialize the digital pin as an output.
        // Pin 13 has an LED connected on most Arduino boards:
        pinMode(DigitalPin.PIN_10, PinMode.OUTPUT);
        pinMode(DigitalPin.PIN_9, PinMode.OUTPUT);
        pinMode(DigitalPin.PIN_8, PinMode.OUTPUT);
        pinMode(DigitalPin.PIN_7, PinMode.OUTPUT);
        pinMode(DigitalPin.PIN_6, PinMode.OUTPUT);
        pinMode(DigitalPin.PIN_5, PinMode.OUTPUT);
        pinMode(DigitalPin.PIN_4, PinMode.OUTPUT);
        pinMode(DigitalPin.PIN_3, PinMode.OUTPUT);
        pinMode(DigitalPin.PIN_2, PinMode.OUTPUT);

        
    }

    public void loop() {
    	   // Schaltet nach der reihe die LEDs auf den Pins an
        digitalWrite(DigitalPin.PIN_10, DigitalState.HIGH);
        delay(1000); // Zeit Einstellung
   
        digitalWrite(DigitalPin.PIN_9, DigitalState.HIGH);
        delay(1000); // Zeit Einstellung
    
        digitalWrite(DigitalPin.PIN_8, DigitalState.HIGH);
        delay(1000); //Zeit Einstellung

        digitalWrite(DigitalPin.PIN_7, DigitalState.HIGH);
        delay(1000); //Zeit Einstellung
    
        digitalWrite(DigitalPin.PIN_6, DigitalState.HIGH);
        delay(1000); // Zeit Einstellung

        digitalWrite(DigitalPin.PIN_5, DigitalState.HIGH);
        delay(1000); // Zeit Einstellung

        digitalWrite(DigitalPin.PIN_4, DigitalState.HIGH);
        delay(1000); //Zeit Einstellung

        digitalWrite(DigitalPin.PIN_3, DigitalState.HIGH);
        delay(1000); // Zeit Einstellung

        digitalWrite(DigitalPin.PIN_2, DigitalState.HIGH);
        delay(1000); // Zeit Einstellung
        
        
        // Schalet LEDs wieder AUS 
        digitalWrite(DigitalPin.PIN_10, DigitalState.LOW);
        delay(1000); // Zeit Einstellung

        digitalWrite(DigitalPin.PIN_9, DigitalState.LOW);
        delay(1000); //Zeit Einstellung

        digitalWrite(DigitalPin.PIN_8, DigitalState.LOW);
        delay(1000); // Zeit Einstellung

        digitalWrite(DigitalPin.PIN_7, DigitalState.LOW);
        delay(1000); //Zeit Einstellung
        
        digitalWrite(DigitalPin.PIN_6, DigitalState.LOW);
        delay(1000); // Zeit Einstellung

        digitalWrite(DigitalPin.PIN_5, DigitalState.LOW);
        delay(1000); // Zeit Einstellung

        digitalWrite(DigitalPin.PIN_4, DigitalState.LOW);
        delay(1000); // Zeit Einstellung

        digitalWrite(DigitalPin.PIN_3, DigitalState.LOW);
        delay(1000); //Zeit Einstellung
 
        digitalWrite(DigitalPin.PIN_2, DigitalState.LOW);
        delay(1000); // Zeit Einstellung
        

        //Beendet die Schleife und das Programm
    }

	public static void main(String[] args) {
        String serialPort;
//        if (args.length == 1) {
            serialPort = "COM7";
//        } else {
//            serialPort = Serial4JArduino.selectSerialPort();
//        }
        JArduino arduino = new Auto_Test_Folge(serialPort);
        arduino.runArduinoProcess();
    }

	}