SINTEF-9012/JArduino #43: Jarduino witht Arduino Mega

frickerm, 2015-12-10:
Hey there 
I want to use Jarduino with an Arduino Mega 
because i need more digital output pin

but if i enter PIN_33 Output for example there is an error 
because they dont know the variable PIN_33

Can i extends the source code and how can i do this

brice-morin, 2015-12-10:
Hi,

Yes, you can extend JArduino to support Arduino Mega. Some files you should extend:
- https://github.com/SINTEF-9012/JArduino/blob/master/jarduino.core/src/main/java/org/sintef/jarduino/AnalogPin.java
- https://github.com/SINTEF-9012/JArduino/blob/master/jarduino.core/src/main/java/org/sintef/jarduino/DigitalPin.java
- https://github.com/SINTEF-9012/JArduino/blob/master/jarduino.core/src/main/java/org/sintef/jarduino/PWMPin.java
- (Maybe) https://github.com/SINTEF-9012/JArduino/blob/master/jarduino.core/src/main/java/org/sintef/jarduino/InterruptPin.java

You might need to extend the JArduino firmware as well, adding pins:
- https://github.com/SINTEF-9012/JArduino/blob/master/jarduino.core/src/main/arduino/JArduino/JArduino.h

Maybe more files, but you can start with that and it should pretty much be ok

frickerm, 2015-12-10:
Ok thank you 
I found and change the JArduino.h file 
but i don´t can find the DigialPin.java 
where was there saved

brice-morin, 2015-12-10:
Not sure I understood... but if you are looking for the DigitalPin.java file, you can just click the link https://github.com/SINTEF-9012/JArduino/blob/master/jarduino.core/src/main/java/org/sintef/jarduino/DigitalPin.java

frickerm, 2015-12-10:
Okay 
I downloaded the JArduino 0.1.7b because i has Updatet my Arduino Software
then i copy the firmware in the sample folder of the Arduino softwar
now i load the other files of the dowload in my java Ide (eclipse)

the import DigitalPin works 
but i dont know where does the import saved 
and where can i change

frickerm, 2015-12-10:
I found them on lib 
but there are all class files how can i change them now

brice-morin, 2015-12-11:
Maybe the easiest is to start from the source, by cloning this repo.

frickerm, 2015-12-11:
okay now i changed the class file 

to this 

/**
 * Licensed under the GNU LESSER GENERAL PUBLIC LICENSE, Version 3, 29 June 2007;
 * you may not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 *
 * 	http://www.gnu.org/licenses/lgpl-3.0.txt
 *
 * Unless required by applicable law or agreed to in writing, software
 * distributed under the License is distributed on an "AS IS" BASIS,
 * WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 * See the License for the specific language governing permissions and
 * limitations under the License.
 *
 * Authors: Franck Fleurey and Brice Morin
 * Company: SINTEF IKT, Oslo, Norway
 * Date: 2011
 */
package org.sintef.jarduino;

import java.util.HashMap;
import java.util.Map;

public enum DigitalPin 
{
	PIN_0((byte)0),
	PIN_1((byte)1),
	PIN_2((byte)2),
	PIN_3((byte)3),
	PIN_4((byte)4),
	PIN_5((byte)5),
	PIN_6((byte)6),
	PIN_7((byte)7),
	PIN_8((byte)8),
	PIN_9((byte)9),
	PIN_10((byte)10),
	PIN_11((byte)11),
	PIN_12((byte)12),
	PIN_13((byte)13),
	A_0((byte)14),
	A_1((byte)15),
	A_2((byte)16),
	A_3((byte)17),
	A_4((byte)18),
	A_5((byte)19);
	PIN_22((byte)22),
	PIN_23((byte)23),
	PIN_24((byte)24),
	PIN_25((byte)25),
	PIN_26((byte)26),
	PIN_27((byte)27),
	PIN_28((byte)28),
	PIN_29((byte)29),
	PIN_30((byte)30),
	PIN_31((byte)31),
	PIN_32((byte)32),
	PIN_33((byte)33),
	PIN_34((byte)34),
	PIN_35((byte)35),
	PIN_36((byte)36),
	PIN_37((byte)37),
	PIN_38((byte)38),
	PIN_39((byte)39),
	PIN_40((byte)40);
	

	private final byte value;
	
	private DigitalPin(byte value)
{
		this.value = value;
}
	
	public byte getValue()
{
		return value;
}
	
	private static final Map<Byte, DigitalPin> map;
	
	static {
		map = new HashMap<Byte, DigitalPin>();
		map.put((byte)0, DigitalPin.PIN_0);
		map.put((byte)1, DigitalPin.PIN_1);
		map.put((byte)2, DigitalPin.PIN_2);
		map.put((byte)3, DigitalPin.PIN_3);
		map.put((byte)4, DigitalPin.PIN_4);
		map.put((byte)5, DigitalPin.PIN_5);
		map.put((byte)6, DigitalPin.PIN_6);
		map.put((byte)7, DigitalPin.PIN_7);
		map.put((byte)8, DigitalPin.PIN_8);
		map.put((byte)9, DigitalPin.PIN_9);
		map.put((byte)10, DigitalPin.PIN_10);
		map.put((byte)11, DigitalPin.PIN_11);
		map.put((byte)12, DigitalPin.PIN_12);
		map.put((byte)13, DigitalPin.PIN_13);

		map.put((byte)22, DigitalPin.PIN_22);
		map.put((byte)23, DigitalPin.PIN_23);
		map.put((byte)24, DigitalPin.PIN_24);
		map.put((byte)25, DigitalPin.PIN_25);
		map.put((byte)26, DigitalPin.PIN_26);
		map.put((byte)27, DigitalPin.PIN_27);
		map.put((byte)28, DigitalPin.PIN_28);
		map.put((byte)29, DigitalPin.PIN_29);
		map.put((byte)30, DigitalPin.PIN_30);
		map.put((byte)31, DigitalPin.PIN_31);
		map.put((byte)32, DigitalPin.PIN_32);
		map.put((byte)33, DigitalPin.PIN_33);
		map.put((byte)34, DigitalPin.PIN_34);
		map.put((byte)35, DigitalPin.PIN_35);
		map.put((byte)36, DigitalPin.PIN_36);
		map.put((byte)37, DigitalPin.PIN_37);
		map.put((byte)38, DigitalPin.PIN_38);
		map.put((byte)39, DigitalPin.PIN_39);
		map.put((byte)40, DigitalPin.PIN_40);
		
		map.put((byte)14, DigitalPin.A_0);
		map.put((byte)15, DigitalPin.A_1);
		map.put((byte)16, DigitalPin.A_2);
		map.put((byte)17, DigitalPin.A_3);
		map.put((byte)18, DigitalPin.A_4);
		map.put((byte)19, DigitalPin.A_5);
	}
	
	public static DigitalPin fromValue(byte b) {
		return map.get(b);
	}
	
}

but eclipse dont fint PIN_40 for example

frickerm, 2015-12-11:
or can you send me the chanced import jar files via email

m.fricker@gmx.de

brice-morin, 2015-12-11:
You can regenerate the jars with Maven. man clean install

frickerm, 2015-12-11:
Ok 
i dont know what i should change 

did someone else changed this and could send me
please

frickerm, 2015-12-14:
when i copy the changed  DigitalPin class file in the jar file (imports)

should i certify the class file 

because if i open the class files with an editor  there are all encoded
and the changed is readable

brice-morin, 2015-12-14:
I would not recommend changing the content of `.jar` files by copy/pasting modified `.class` files.

- install Maven
- modify the sources as you did
- run mvn clean install at the root

frickerm, 2015-12-16:
sorry but it dont work

ScottBruton, 2017-06-05:
Guys, I really want to extend jArduino to work with Arduino Mega.
I tried understanding the above threads but I'm still unsure what to do to do this.

Please can someone tell me step by step how to change jArduino files to be able to include all Analog Pins ( A0 - A15) and Digital pins (D22-D53).

Thats all I need.

Thank you
Scott