SINTEF-9012/JArduino #49: java.lang.IllegalArgumentException in line 42 in AbstractJArduino.java

lobsteroh, 2016-03-15:
line 42 in AbstractJArduino.java

    serial = (JArduinoClientObserver) clazz.getConstructor(String.class).newInstance(ID,conf);

throws ...

    java.lang.IllegalArgumentException: wrong number of arguments
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at org.sintef.jarduino.AbstractJArduino.<init>(AbstractJArduino.java:42)
	at org.sintef.jarduino.JArduino.<init>(JArduino.java:48)
	at Blink_RH.<init>(Blink_RH.java:37)
	at Blink_RH.main(Blink_RH.java:65)
    Exception in thread "Thread-4" java.lang.NullPointerException
	at org.sintef.jarduino.AbstractJArduino.pinMode(AbstractJArduino.java:64)
	at Blink_RH.setup(Blink_RH.java:41)
	at org.sintef.jarduino.JArduino$RemoteArduinoProcess.run(JArduino.java:132)

when i 'mvn clean install' Arduino/jarduino.core/pom.xml to recompile org.sintef.jarduino.core-0.1.7-SNAPSHOT,jar and replace the copy that is in the distribution for examples 0.1.7b, then run Blink.java or any other example.

brice-morin, 2016-03-15:
I just tried `mvn clean install` from the root (which will also build jarduino.core) and it worked... Did you recompile from the sources you cloned from GitHub?

lobsteroh, 2016-03-15:
i was not clear, tried to shorten, failed miserably ... yes, 'mvn clean install' builds with success and creates org.sintef.jarduino.core-0.1.7-SNAPSHOT,jar. i use it to replace the version of that jar that is included in the newest distribution for 0.1.7b and run example Blink ... or any other example in the list. All of them fail at runtime while instantiating the basic JArduino object.

I need to compile the jars from source as I want to define the additional pins from the Mega 2560.

brice-morin, 2016-03-15:
OK, it should work now. Just recompile the whole project. Tell me if you still have issues

closed 2016-03-15 by brice-morin

lobsteroh, 2016-03-15:
Hi Brice, quick note to confirm that this fixed the issue. much obliged. Robert

commit ca185954 (2016-03-15): fix #49 (hopefully)