SINTEF-9012/JArduino #53: firmware not functional when compiled from source

lobsteroh, 2016-04-11:
took me a while to tack down this issue ... osx 10.11.4

JArduino firmware upload fails with 

    /Users/lobsterman/Documents/C_Arduino/Arduino/libraries/JArduino/examples/JArduinoFirmware/JArduinoFirmware.pde: In function 'void setup()':
    JArduinoFirmware:87: error: no matching function for call to 'JArduino::init_JArduino()'

this only happens if i clone

    https://github.com/SINTEF-9012/JArduino.git

then cd into it and mvn clean install, then copy 

    ~/Documents/workspace/JArduino/jarduino.core/src/main/arduino into my Arduino libraries folder

however, if i use the JArduinoFirmware that comes with the binary distribution JArduino-0.1.7b.zip, then it works just fine.

brice-morin, 2016-04-11:
OK, I suspect the problem to come from with default value on parameter:
```
void JArduino::init_JArduino(uint16_t baud = 9600) {
	// init the serial port
	Serial.begin(baud);
}
```

Try replacing with:
```
void JArduino::init_JArduino(void) {
	// init the serial port
	Serial.begin(9600);
}
```

lobsteroh, 2016-04-11:
yes, confirming that fixing this in JArduino.cpp within Arduino/libraries makes the firmware work. thanks

closed 2016-04-11 by lobsteroh