SINTEF-9012/JArduino #54: analogRead in JArduino reports incorrect range of values

lobsteroh, 2016-04-11:
same hardware setup (4 channel capacitance sensor OSEPP 4DTOUCH-01. channel 1 into analog pin A0) reports very different values and ranges when running AnalogInOutSerial.java or built-in example AnalogInOutSerial 

Arduino example gives background values around 350, increasing to 900 when I touch sensor
JArduino example gives background values around 350, increasing to 385 when i touch sensor

i suspect another error related to unsigned byte vs java byte (8-bit signed) issue i had flagged earlier

in java 
(byte)129 = (numeric) -128
(byte)255 = (numeric)-1
while (byte)0 - (byte)128 give numeric 0-128

brice-morin, 2016-04-11:
Yes, it is probably related to signed/unsigned byte. We might use something like Google's Guava, which support [UnsignedBytes](http://docs.guava-libraries.googlecode.com/git/javadoc/com/google/common/primitives/UnsignedBytes.html) in Java. I am however uncertain at this point how big of a refactoring that would imply.

lobsteroh, 2016-04-11:
blush ... i was tracking down exactly where the byte[2] gets stuffed into the short with the plan of intercepting it with a call to Guava. i was following the flow of bytes with echo to console when ... the thing starts working AS IS, i can only imagine that one of my jumpers had missed making a connection properly. so, all goo. it was a false alarm ... blushing again ...

closed 2016-04-11 by lobsteroh

brice-morin, 2016-04-11:
Ok, no problem :-)