SINTEF-9012/JArduino #55: DigitalState.HIGH is not outputting correct value equivalent of (byte)127

lobsteroh, 2016-04-13:
creating a DigitalWriteMsg(DigitalPin pin, DigitalState value) with

    digitalWrite(DigitalPin.PIN_11,DigitalState.HIGH);

is not turning on 5V - powering an led makes it only blink very dimly

in contrast using writing analog to PWM with

    analogWrite(PWMPin.PWM_PIN_11,(byte)127)

makes it light up like a daemon. actually powering the PWMPin with (byte)1 gives the same brightness as DigitalState.HIGH.

i am thinking that writing DigitalState.HIGH to the DigitalPin ought to send (byte)127. several ways I can think of doing this ...

1. obvious fix - defining DigitalState.HIGH with 127 (or 255 for that matter works as well although it runs into the signed/unsigned byte issue) instead of 1 in DigitalState.java

2. creating a DigitalWriteMsg with value 127 when DigitalState.HIGH is provided

any prefs?

brice-morin, 2016-04-13:
Yes, this issue is the same as #44 

I guess I will go with option one and change [this line](https://github.com/SINTEF-9012/JArduino/blob/master/jarduino.core/src/main/java/org/sintef/jarduino/DigitalState.java#L25)

brice-morin, 2016-04-13:
Hopefully it should work now

lobsteroh, 2016-04-13:
looks like i need to create and send low level DigitalWriteMsg to figure out exactly where the thing goes wrong ... still only getting 2.5V on DigitalPin 8 with DigitalState.HIGH

brice-morin, 2016-04-13:
Ok... Did you properly set the mode of that pin before setting it to High?

lobsteroh, 2016-04-13:
Hi Brice, I understand that you want to include the type checking by declaring enums but it realy limits testing when the behavior is not what it is meant to be. on my copy of JArduino i have added the constructor 

	public DigitalWriteMsg(DigitalPin pin, byte byteValue) {
		setCommandID(JArduinoProtocol.DIGITAL_WRITE);
		setByteValue(pin.getValue());
		this.pin = pin;
		setByteValue(new Byte(byteValue));
                // this.value = value;
	}

and in AbstractJArduino

    public void digitalWrite(DigitalPin pin, byte value) {
        // Create message using the factory
        FixedSizePacket p = JArduinoProtocol.createDigitalWrite(pin, value);
        // Send the message on the serial line
        serial.receiveMsg(p.getPacket());
    }
so now i can send specific byte values to test digitalWrite(). This is the test class i am using where you can just comment 

    import org.sintef.jarduino.DigitalPin;
import org.sintef.jarduino.JArduino;
import org.sintef.jarduino.PWMPin;

public class ArduinoTestWrite extends JArduino {

	//static Object outPin = PWMPin.PWM_PIN_11;
	static Object outPin = DigitalPin.PIN_11;

	static byte onVal = (byte)127;
	
	public ArduinoTestWrite(String port) { super(port); }

	@Override
	protected void setup() {}

	@Override
	protected void loop() {}

	public static void main(String[] args) {
		try {
			String serialPort = "/dev/tty.usbmodem1421";
			ArduinoTestWrite arduino = new ArduinoTestWrite(serialPort);
			arduino.runArduinoProcess();
			Thread.sleep(2000);
			
			if (outPin instanceof PWMPin)
				arduino.analogWrite((PWMPin)outPin,onVal);
			else if (outPin instanceof DigitalPin)
				arduino.digitalWrite((DigitalPin)outPin,onVal);
			System.out.println("Turning on LED with byte value: " + onVal);
			Thread.sleep(1000);
			
			arduino.stopArduinoProcess();
		} catch (Exception e) { e.printStackTrace(); }
	}
    }

Switching PWMPin to the DigitalPin line reduces output to half. I am puzzled where the problem sits ... suggestions?

lobsteroh, 2016-04-13:
you got it. yes, setting pinMode makes all the difference, but it has to happen after the arduino is listening. this version now works ...

    import org.sintef.jarduino.DigitalPin;
    import org.sintef.jarduino.JArduino;
    import org.sintef.jarduino.PWMPin;
    import org.sintef.jarduino.PinMode;

    public class ArduinoTestWrite extends JArduino {

	//static Object outPin = PWMPin.PWM_PIN_11;
	static Object outPin = DigitalPin.PIN_11;

	static byte onVal = (byte)127;
	
	public ArduinoTestWrite(String port) { 
		super(port); 
	}

	@Override
	protected void setup() {}

	@Override
	protected void loop() {}

	public static void main(String[] args) {
		try {
			String serialPort = "/dev/tty.usbmodem1421";
			ArduinoTestWrite arduino = new ArduinoTestWrite(serialPort);
			arduino.runArduinoProcess();
			Thread.sleep(2000);
			
			if (outPin instanceof PWMPin)
				arduino.analogWrite((PWMPin)outPin,onVal);
			else if (outPin instanceof DigitalPin) {
				arduino.pinMode((DigitalPin)outPin, PinMode.OUTPUT);
				arduino.digitalWrite((DigitalPin)outPin,onVal);
			}
			System.out.println("Turning on LED with byte value: " + onVal);
			Thread.sleep(1000);
			
			arduino.stopArduinoProcess();
		} catch (Exception e) { e.printStackTrace(); }
	}
    }

commit f6f5c7e6 (2016-04-13): fix #55, fix #44
closed 2016-04-13 by lobsteroh