SINTEF-9012/JArduino #56: initialize jArduino in firmware - not supported

lobsteroh, 2016-04-14:
recommendation for supporting higher baud rate in firmware is via ...

    void setup() {
	_JArduino.init_JArduino(115200);
    }

unfortunately uploading this to arduino produces with compile from source gives ...

    exit status 1
    no matching function for call to 'JArduino::init_JArduino(long int)'

brice-morin, 2016-04-14:
OK, I guess we should change the signature of [that method](https://github.com/SINTEF-9012/JArduino/blob/master/jarduino.core/src/main/arduino/JArduino/JArduino.cpp#L36) so that it can take a long int as parameter

lobsteroh, 2016-04-14:
yeah, i have been playing in changing the .h and .cpp files for having both constructors. in header ...

    	void init_JArduino();
 	void init_JArduino(uint16_t baud);

and .cpp

    // Public Methods //////////////////////////////////////////////////////////////
    void JArduino::init_JArduino(uint16_t baud = 9600) {
	// init the serial port
	Serial.begin(baud);
    }

    void JArduino::init_JArduino(void) {
	// init the serial port
	init_JArduino(9600);
    }

but upload of the new firmware to arduino fails ... what am i missing?

lobsteroh, 2016-04-14:
ok, in the firmware i try ...

    void setup() {
	uint16_t baud = 9600;
        _JArduino.init_JArduino(baud);
    }

and that works, but how to pass in the baud rate from jArduino?

closed 2016-05-08 by lobsteroh