SINTEF-9012/JArduino #57: timing an analog read ...

lobsteroh, 2016-04-18:
reading analog pins in jArduino appears, well, glacial ....
when i time 1000 reads with this

    int counter = 0;
    int startT = millis();
    while (counter<1000) {
        sensorValue = analogRead(sensorPin);
        counter++;
    }
    int durT = millis() - startT;

Arduino via IDE reports 112ms
JArduino does it in, wait for it, ... 40970ms :-( 
- 41ms per read? WTF?

am i missing something? if true, this would relegate jArduino to a niche market, at best ...

aliassad, 2016-04-18:
Did you use JArduino AdvanceGUI ?

brice-morin, 2016-04-18:
Well, I will not even think about starting to potentially investigate on that issue before it has been edited and turned into a less aggressive style. Had you read the wiki, it explained some tradeoffs about JArduino (though things can certainly be improved).

lobsteroh, 2016-04-19:
Hi Brice, apologies, clumsy tone ment to express my shocked surprise rather
than any adversial sentiment. I had hoped that arduinos would be suited to
address my i/o as well as my i2c needs. On paper it should offer a cheap
and user-friendly solution, alas, my conclusion after giving it a fair
shake is that the platform may not be there. A mention or reference to
design intentions in the readme would be welcome. Ended up switching to
linkm hardware for I2c and phidgets for everything else. Java drivers are
quite straight forward in these and interfaces perform well. Got my project
solved in one weekend. Still glad to have figured capabilities and limits
of arduino platform. Thanks for your help.

-- 

Robert
------------------------------------------------------------------

Robert Huber, PhD
Helen Putnam Fellow, Radcliffe Institute for Advanced Study
Harvard University, Cambridge, MA 02138
Professor, Biological Sciences,  BGSU
Associate Editor - Acta Ethologica
Associate Editor - Frontiers in Physiology
Email: rh.bgsu@gmail.com

------------------------------------------------------------------

closed 2016-05-08 by lobsteroh