SINTEF-9012/JArduino #60: java.lang.ClassNotFoundException: org.sintef.jarduino.comm.Serial4JArduino

SkyHawkBre, 2016-05-19:
I am tring to run the Blink example:
`import org.sintef.jarduino.*;

public class Blink extends JArduino 
{

	public Blink(String port) 
	{
	    super(port);
	}
	
	@Override
	protected void setup() 
	{
	    // initialize the digital pin as an output.
	    // Pin 13 has an LED connected on most Arduino boards:
	    pinMode(DigitalPin.PIN_12, PinMode.OUTPUT);
	}
	
	@Override
	protected void loop() 
	{
		
	    // set the LED on
	    digitalWrite(DigitalPin.PIN_12, DigitalState.HIGH);
	    delay(1000); // wait for a second
	    // set the LED off
	    digitalWrite(DigitalPin.PIN_12, DigitalState.LOW);
	    delay(1000); // wait for a second
	}
	
	public static void main(String[] args) 
	{
	    String serialPort = "/dev/cu.wchusbserial1410";
	    JArduino arduino = new Blink(serialPort);
	    arduino.runArduinoProcess();
	}
}`
And I am getting the following error:
`java.lang.ClassNotFoundException: org.sintef.jarduino.comm.Serial4JArduino
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:331)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	at org.sintef.jarduino.AbstractJArduino.<init>(AbstractJArduino.java:41)
	at org.sintef.jarduino.JArduino.<init>(JArduino.java:48)
	at Blink.<init>(Blink.java:8)
	at Blink.main(Blink.java:34)
Exception in thread "Thread-0" java.lang.NullPointerException
	at org.sintef.jarduino.AbstractJArduino.pinMode(AbstractJArduino.java:64)
	at Blink.setup(Blink.java:16)
	at org.sintef.jarduino.JArduino$RemoteArduinoProcess.run(JArduino.java:128)`

I am new to GitHub and don't know what I am doing wrong, any help is greatly appreciated :)
-Sky

MrLuiso, 2016-10-20:
I'm having the same problem.

jsutaria, 2016-10-21:
Also having the same problem! Any thoughts?

MrLuiso, 2016-10-25:
I found the missing class on the master folder. If you do a quick search it
will come up. Now all I have to do is resolve an issue with an import.

gnu.io.* to be more specific