SINTEF-9012/JArduino #69: define PIN schematics for different arduino models

lobsteroh, 2016-11-17:
I was wondering whether and, if so, how, JArduino handles the different pinouts for different arduino models. answers on this in past issues didn't seem conclusive. the rather rigid definition of pins in JArduino is great when it prevents us from using digital pins for analog writes, however, it also really limits our ability to use boards that have different pin characteristics. as far as i can tell only the uno is defined. the mega is the more useful board because it controls so many pins at a great price. i can change the pin definitions and recompile with maven but that will get mega working while it breaks the uno. i suspect separate pinout schematics for different models are the way to go but i dont see whether that has been tackled. any suggestions or advice on preferred design?

brice-morin, 2016-11-17:
One problem is that Java does not allow for extending enumerations. One possible way of doing it is to define an abstract class `Platform` and then have two (or more) concrete classes `UNO` extending `Platform` and `MEGA` extending `Platform` or `UNO`. Each concrete class can define a set of `static final PIN` element. We should then be able to use pins like this: `UNO.PIN1` or `MEGA.PIN1`.

lobsteroh, 2016-11-17:
yeah, not a bad tack. could you be tempted to take a stab at the basic layout of classes? i'd be happy to add pin schematic details for individual models, if you like. teensie would be another model to include, by far the best hardware of the bunch in my hands. Alternatively, a much simpler solution would be to drop the rigid link between pin function and address, but, i guess, you would not be a friend of that.

brice-morin, 2016-11-17:
I am quite busy at work until end of November and cannot unfortunately commit to look into this issue in more details before early December. I suspect it is just a 20-minute refactoring, but you never know before it is done...