SINTEF-9012/JArduino #83: Cant install your library

Munoon, 2019-04-06:
Hi! Im trying to install your library but can't. I didn't find any tutorial about installing your library so i install it from mvnrepository.com. Im installed all your three librarys (https://mvnrepository.com/search?q=org.sintef.jarduino) but when i compile code i have an error:
`Exception in thread "main" java.lang.NoClassDefFoundError: gnu/io/SerialPortEventListener
	at org.sintef.jarduino.AbstractJArduino.<init>(AbstractJArduino.java:30)
	at org.sintef.jarduino.JArduino.<init>(JArduino.java:45)
	at Blink.<init>(Blink.java:8)
	at Blink.main(Blink.java:25)
Caused by: java.lang.ClassNotFoundException: gnu.io.SerialPortEventListener
	at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:583)
	at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:178)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:521)
	... 4 more`

brice-morin, 2019-04-06:
Can you try cloning the repo and run `mvn clean install` in the JArduino folder?

Munoon, 2019-04-07:
yeap, and what?

Munoon, 2019-04-07:
But when i try that
`$ cd jarduino.samples
$ mvn exec:java -Dexec.mainClass="org.sintef.jarduino.gui.JArduinoGUI"`
I still have an error in console
`
java.lang.ClassNotFoundException: org.sintef.jarduino.gui.JArduinoGUI                                                  at java.net.URLClassLoader.findClass (URLClassLoader.java:471)                                                     at java.lang.ClassLoader.loadClass (ClassLoader.java:588)                                                          at java.lang.ClassLoader.loadClass (ClassLoader.java:521)                                                          at org.codehaus.mojo.exec.ExecJavaMojo$1.run (ExecJavaMojo.java:270)                                               at java.lang.Thread.run (Thread.java:834)
--

Munoon, 2019-04-29:
so may someone help me pls?

brice-morin, 2019-04-29:
Retry `mvn clean install` at the root folder after uncommenting [this line](https://github.com/SINTEF-9012/JArduino/blob/master/pom.xml#L33)

closed 2019-07-24 by Munoon