SINTEF-9012/JArduino #86: "Load RxTx" error

Soloboulou, 2019-12-25:
Hello, 

I am trying to install JArduino by following the 2 minutes tutorial then executing a "Blink" program. 

I have no error in my "JArduino" projects before running, but then, it says "Load RxTx". 

I have tried to place librxtxserial.jnilib (I have a Macbook) and RxTxComm everywhere (in Java/Extension, in Referrenced Librairies, in my project folder, ... ). It never works. 

Any ideas ? 

Thank you, 
Happy Christmas !

warneobyte, 2019-12-31:
Just Need download the RTXT .dll files from here http://fizzed.com/oss/rxtx-for-java .inside the zip follow the steps for install , it worked for me .

Soloboulou, 2020-01-04:
Hello, 

It didn't worked for me, maybe because I have a Mac ? 

I retried everything, put the rxtx.dll + jnilib + rxtxcommport in my project, now when I run Blink, I have this error : 

librxtxSerial.jnilib: no matching architecture in universal wrapper thrown while loading gnu.io.RXTXCommDriver

Any idea ? 

Thank you !